JaredC01/LabStack #10: module suggestion

ConanWolf46, 2025-01-17:
https://www.amazon.com/dp/B01J705MTC?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1

this amazon product is a sim data to USB for internet via phone service, there is a version of the product on aliexpress as well however i only linking amazon because idk how people feel about aliexpress on here, the version on aliexpress is 9 USD. 

the module idea that i was having was similarly to the antenna coming out of the back but for the device, the USB port on the device is in the front (antenna is in the back) making it perfect for that 4 hole module setup that you have already

JaredC01, 2025-01-17:
> https://www.amazon.com/dp/B01J705MTC?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1 just a fair warning about this product listing, the PCIE module is not included
> 
> this amazon product is a sim data to USB for internet via phone service, there is a version of the product on aliexpress as well however i only linking amazon because idk how people feel about aliexpress on here, the version on aliexpress is 9 USD.
> 
> the module idea that i was having was similarly to the antenna coming out of the back but for the device, the USB port on the device is in the front (antenna is in the back) making it perfect for that 4 hole module setup that you have already

Just so I'm clear, you're looking for a way to mount this device in a module at the front of the LabStack, correct?  Are you wanting to break out the USB port as well via Keystone Jack?

bemany, 2025-01-21:
Request for Intel Nuc BOXNUC8i5BEK2

I am not able to find a simple solution for mounting the nuc on its mounting screws on a 10" module

[image]
https://www.intel.com/content/www/us/en/support/articles/000006857/intel-nuc/intel-nuc-kits.html

It would also be nice to ziptie the power adapter next to it "huntkey HKA09019047-6u"
https://www.ebay.de/itm/256636100141

Break out of HDMI, Keystone .. is not mandatory

ConanWolf46, 2025-02-16:
hay sorry for not responding for a bit, ideally yes because this device requires a USB connection and the goal is for a raspberry pi server over a phone network

ConanWolf46, 2025-02-16:
after the pi gets the network via the device, i could if i wanted to split the network off to multiple other pis if needed or just have the one pi as needed.