JaredC01/LabStack #12: Module for RPI 1

oNutti, 2025-01-17:
Hello, 
I currently run RPI 4, RPI 1 and a switch, and half a patch panel. This project is perfect for me, i would just love to have a mount for RPI 1.

JaredC01, 2025-01-17:
Does the RPi 1 not use the same mount as the rest of the Pi's?  I'm not familiar with the older Pi's, but I thought they shared the same form factor as the newer ones.  A single 2x Pi mount should fit both your Pi 4 and your Pi 1 (with 3 modules open for other things).

oNutti, 2025-01-17:
> Does the RPi 1 not use the same mount as the rest of the Pi's? I'm not familiar with the older Pi's, but I thought they shared the same form factor as the newer ones. A single 2x Pi mount should fit both your Pi 4 and your Pi 1 (with 3 modules open for other things).

Unfortunately, not. The RPI 1 (maybe 2?) uses other mounting holes, actually only two. The 3, 4, 5 (maybe 2) uses the same holes. It's one of the dumbest things, they made..

oNutti, 2025-01-17:
Yes, the RPI 1 is can't be more outdated, but i still use it for hosting local controllers - someone other than me, might also use it, haha.

JaredC01, 2025-01-17:
I'm guessing it's more to do with the model "B" being the same (it looks like there's a Pi 1 Model B that looks to be the same as the rest). Do you have a picture of the Pi 1 you have for reference to know what I'm looking for?

oNutti, 2025-01-17:
The RPI 1 model B+ use the same mounting holes as the other ones, but funny enough does the RPI 1 use some other ones.

[image]

JaredC01, 2025-01-17:
Too many letters and pluses! 🤣

I should be able to find a model for that Pi 1 and add a mount for it, but I don't expect I'll be able to make it work with a dual-Pi mount (meaning it would be a single module unit). So long as that's not an issue, I'll add it to the list!

oNutti, 2025-01-17:
> Too many letters and pluses! 🤣
> 
> I should be able to find a model for that Pi 1 and add a mount for it, but I don't expect I'll be able to make it work with a dual-Pi mount (meaning it would be a single module unit). So long as that's not an issue, I'll add it to the list!

Not a problem. I only use two atm, and dont plan on more than 3, so perfect!
Thanks!

oNutti, 2025-01-18:
Hello again,
Just wanted to mention, that the RPI 1 also use the big SD card, and not the microSD cards. 
Thanks for this nice project, once again!

oNutti, 2025-01-28:
Hey, 
Just wanted a status. Any updates?

JaredC01, 2025-01-28:
Still on the list!  I've been sorting out some of the other modules that didn't need a full-custom design.  I may have some time tomorrow to get it sorted, otherwise it will likely be Thursday or Friday.

oNutti, 2025-01-28:
Sounds perfect!

JaredC01, 2025-01-30:
Okay, I've got a "preliminary" module created... It's specifically for the Pi 1 Model B Rev 2.0 (which is the version of the Pi 1 B that has mounting holes, apparently).  Please let me know if it works!

https://github.com/JaredC01/LabStack/blob/main/Modules/STLs/Module%201x%20Pi%201B%20Rev%202.0.stl

oNutti, 2025-01-30:
Perfect,
Printer is busy for the next hours. But i will write a review later
Many thanks

JaredC01, 2025-01-30:
> This RPI model is the most insane ever.
> 
> [image]
> On the USB ports there are small things on the side. No idea what to call them.
> 
> I just filed down the usb port on the print and it worked. A fix could be, making the usb port bigger on every side. Maybe with half a milimeter on every side. 

I'll add another millimeter on the top and bottom for the little "clamps". Can you check with something plugged in to see if they go out further than 1mm on either side?

Other than that does it fit?

oNutti, 2025-01-30:
It fit perfectly. For you information, it would require 1 milimeter on every side. 
" Can you check with something plugged in to see if they go out further than 1mm on either side?", not sure what you mean with this. Plugging in a usb or two, doesn't do anything to the mount at all.

JaredC01, 2025-01-30:
> It fit perfectly. For you information, it would require 1 milimeter on every side. " Can you check with something plugged in to see if they go out further than 1mm on either side?", not sure what you mean with this. Plugging in a usb or two, doesn't do anything to the mount at all.

Oh you mean the "winglets" on the USB plug itself.  I thought you meant the little spring metal pieces on both the top/bottom and sides and helps hold the plug when you plug in the USB port (visible in your picture above).  I'll add some clearance there on the sides, and see about chamfering the ethernet port a bit for helping with inserting/removing ethernet.

oNutti, 2025-01-30:
Github was down.. 😅
I will reprint it - my dad got a RPI 1, he can get it installed into my rack then

oNutti, 2025-01-30:
[image]
My now fully completed rack (lying, always things to upgrade..)
Second prototype works perfect.

JaredC01, 2025-01-31:
Awesome!  Thank you for the testing and feedback, I'll close the issue!

closed 2025-01-31 by JaredC01