JaredC01/LabStack #3: STEP files please!

sirtimbly, 2025-01-17:
It's much easier to integrate these into my own designs if I have step files to import instead of STLs. Thanks! I'm looking forward to playing with this.

JaredC01, 2025-01-17:
STEP and F3D files will be available soon, I just need to clean up the CAD!  I should have an update this weekend with any luck.

Astrobirder, 2025-01-17:
Thank you ahead of time for the F3Ds & STEP formats!

JaredC01, 2025-01-18:
I've added the F3D CAD file for now.  The STEP file I exported was 180MB due to what I assume are the Pi and Radxa board models, so it's not able to be uploaded to GitHub without moving to a LFS repo (which I don't need to do otherwise).  I'll work on exporting another version without the board models later tonight.

JaredC01, 2025-01-31:
Updated F3D file uploaded, and compressed STEP file uploaded.  I simplified all of the Pi and board models as much as possible, and still ended up with a 230MB STEP file.  Fortunately STEP files are text-based, so I just ZIP'd the STEP to compress it.

closed 2025-01-31 by JaredC01