JaredC01/LabStack #5: Module STL for 1x Pi + 2.5 SSD

DocBrown101, 2025-01-17:
Great project!

Could we get a module for 1x Pi & 2.5 SSD please?

JaredC01, 2025-01-17:
Jeff and I talked about a standalone 2.5" drive module, which is on the roadmap.  I can definitely make a Pi + SSD option, though it will have a tight bend radius if you're tying to install a USB to SATA adapter in the same slot as the Pi next to it.  So long as you're aware of the limitations, I don't mind adding it!

DocBrown101, 2025-01-17:
That would be really great. Thank you!

JaredC01, 2025-01-19:
I've added a Pi + 2.5" SSD option, available here: https://github.com/JaredC01/LabStack/blob/main/Modules/STLs/Module%201x%20Pi%20with%2025%20SSD.stl

I haven't tested it yet, but it should work without issue.  I'll print one once a printer frees up for testing, but please test it on your end as well!

> That would be really great. Thank you!

closed 2025-01-19 by JaredC01

JaredC01, 2025-01-20:
Update: Tested and working!