JaredC01/LabStack #6: Feature request

monkeymademe, 2025-01-17:
Awesome project! I am looking forward to put this into my DeskMate

Would it be possible to have a broken down 2U Rack plate.... My poor Ultimaker 3 is falling a little short 

[image]

JaredC01, 2025-01-17:
You can try splitting the model in the slicer, but I also just got an idea on how I can split the front plate, then reinforce it with the rear braces, in a 4-piece option (left half, right half, 2x braces).  It may need some glue, but the heatsets should act as a means of attachment as well.

I'll add it to my list!

monkeymademe, 2025-01-17:
Awesome thanks that would be helpful (also sorry I forgot to finish the title to this request)

JaredC01, 2025-01-18:
I've added 2 new files located here: https://github.com/JaredC01/LabStack/tree/main/LabStack%20Mini/STLs

The files are labeled as 2U Mini Split Printed Bracket Brace x2, and 2U Mini Split Printed Bracket x2.  You will (as the name implies) print 2 of each of those.

For assembly, line up the split halves, then attach the two brackets over the locating pins.  With the brackets on the locating pins, install the Heatset inserts for each of the screw holes.  There's enough overlap that you should get good support, but you might use some CA glue (super glue) between the brackets and the braces just as a precaution.

Please let me know how printing and assembly goes; you're the guinea pig for testing!

markafarrell, 2025-01-21:
Fyi I printed this out on an Ender 3 and it worked fine. 

It could probably use a little clearance between the two halves as they interfered slightly causing the piece to bow upwards. I just clamped it down between two pieces of wood while the CA glue set and it looks fine. 

[image]

JaredC01, 2025-01-21:
I had added some clearance between them, but it looks like that somehow worked its way back out of the final version (I probably reverted something that broke it).

I've added clearance back (total of 0.4mm, or 0.2mm per side) so it SHOULD be good to go at this point!

markafarrell, 2025-01-21:
Printing the revised version now. Will report back when it's done.

markafarrell, 2025-01-21:
Another thing that might be nice is 4 alignment pins per half instead of 2. I think it will make it a bit easier to make sure things are square when assembling

markafarrell, 2025-01-21:
The STL also appears to be flipped now (alignment pins pointing down) which caught me out cause it needed to be flipped in the slicer

JaredC01, 2025-01-21:
> The STL also appears to be flipped now (alignment pins pointing down) which caught me out cause it needed to be flipped in the slicer

Fixed and pushed.  As far as locating pins goes, two was what fit without running into issues... I can probably add one more set of pins right in the middle where they meet, so each half uses half of the hole.  I'll see what that looks like in CAD.

markafarrell, 2025-01-21:
I was thinking something like. 

[image]

JaredC01, 2025-01-21:
That would end up pretty severely compromising the integrity of the braces, and since they're only used as guides, I don't think the tradeoff is worth it there.

markafarrell, 2025-01-21:
An alternative would be to have one of the pins on the brace instead of a hole.

markafarrell, 2025-01-21:
Actually I think then you wouldn't be able to print the tapered ends of the braces

markafarrell, 2025-01-21:
P.s. the revised version with more clearance works great

JaredC01, 2025-01-24:
> P.s. the revised version with more clearance works great

Good deal!  As far as the extra pins goes, I looked at adding the middle pins that I had talked about, and I don't like it as it leaves too much wiggle room with the gap between the parts.  The heatset inserts will help align the parts, along with the pins, so I think I would prefer to leave it as-is until there's feedback from others.

I'll leave the issue open until I hear a bit more feedback.