JaredC01/LabStack #7: Module for Zimaboard

koreyrebello, 2025-01-17:
Great work - thank you for putting this all together! Any possibility of creating a module for a Zimaboard 832?

JaredC01, 2025-01-17:
Not likely able to do a module for the Zimaboard due to the width of the unit. The spec width is 81mm from what I've been told, while the opening in the LabStack Mini is only 71mm.

I don't have a Zimaboard here to test / verify fit, but if you're able to print the 2U bracket and check fitment, please let me know. If it DOES somehow fit, I certainly don't mind making a module!

I've also proposed a 1U enclosure option that would incorporate space for the PCIe slot and maybe a JetKVM here: https://github.com/JaredC01/LabStack/issues/2

ospfv11, 2025-01-17:
Thanks for the quick response @JaredC01 ! I will take a look at the measurements.