JaredC01/LabStack #8: Any plans for 19-inch racks?

itsCARLsanity, 2025-01-17:
Hey Jared, 

I just saw Jeff's video, amazing work you got there. You got any plans to extending the bracket or whatever the modules are inserted into, to 19-inch racks? Possibly by modularising it by segments and using some kind of connection in between. 

Thanks and have a good day!

JaredC01, 2025-01-17:
Yes! That's also why the repo is LabStack instead of just LabStack Mini. 😃

The 19" side I'll need to test to see if 3D printed parts will hold up well enough or if it needs laser-cut metal parts. Printing a full 19" 2U bracket would also take a pretty large bed, which rules out a lot of cheaper printers, so I'm leaning towards laser-cut metal (at least for the bracket).

itsCARLsanity, 2025-01-17:
That's good to hear! I initially thought its only for 10-inch racks since its primarily featured on Jeff's minilab video. Also, on his site it says "Labstack mini"

I would say, laser-cut metal (for the bracket) for 19-inch racks would limit people from trying out Labstack since  not everybody would have that kind of equipment or it could be pretty expensive. I think the 19 inch 3D printed bracket part would still be worth a while, at least for some it would be an option for them to try out.

JaredC01, 2025-01-17:
> That's good to hear! I initially thought its only for 10-inch racks since its primarily featured on Jeff's minilab video. Also, on his site it says "Labstack mini"
>
> I would say, laser-cut metal (for [...]

The 10" mini rack model is indeed called the LabStack Mini, so that makes sense!  The Mini is also where Jeff and I started with it for his specific mini rack project(s), but it's certainly not the end. 😃

For the laser-cut parts, you wouldn't be making those at home, you'd use a service like https://sendcutsend.com to have them made for a fee, then print the modules at home. The fee for a couple of the brackets from SendCutSend would likely be under $50 USD shipped, so for a project where you're putting in up to 10-12 Raspberry Pi's costing upwards of $500 or more total before you include any HATs, it doesn't seem unreasonable to spend for the laser-cut bracket(s).

Just to be clear, I'm not fully opposed to a printed option for the main bracket!  The concern is whether or not a printed bracket will be rigid enough to support the weight, because a fully-loaded 19" LabStack unit can be heavier than you would think.

I will definitely do some testing, but I fully expect that a printed bracket will not work well for the 19" version of LabStack. If that turns out not to be the case, I'll let you know!

JaredC01, 2025-01-17:
I just put together a 19" rack version of the main LabStack bracket; the total cost at Send Cut Send for two of them is $49, shipped.

[image]

This is obviously only applicable to the US, but does align with what my estimate was in my previous post.

eliasstepanik, 2025-01-21:
Just created a pull request for a 3D Printed version.

itsCARLsanity, 2025-01-26:
Hello~ sorry for late reply. 

> For the laser-cut parts, you wouldn't be making those at home, you'd use a service like https://sendcutsend.com/ to have them made for a fee, then print the modules at home.

Lol. I thought you'd need a laser machine to do it, never knew there's a service for laser cutting metal. This is great then, although for me, I'd be paying more for shipping to get it from the US since I don't know if there's any kind of laser cutting service available from the country I'm in. 

This is awesome work Jared. Thank you!

JaredC01, 2025-01-31:
Worth taking a look to see if you have a local laser-cutting company!  Closing the issue as resolved for now, but let me know if you need anything else!

closed 2025-01-31 by JaredC01