JaredC01/LabStack #9: Contribute Picamera mount (single and double) and hyperpixel display mount

monkeymademe, 2025-01-17:
Hey there,

Again awesome project I also made some Rackmate mounts I would love to share with you if you would like them (or tips for improvement)

Its a little niche for cameras and display but I am sure someone is gonna need it 

Files can be found here (don't know if you accept merge requests): https://github.com/monkeymademe/deskpi-rackmate-prints

Here it is in action 

[image]

JaredC01, 2025-01-17:
Hey there!  Thanks for the link!  I happen to have a Pimoroni HyperPixel 4.0 sitting right next to me that I'm not doing anything with, so I may just have to find a use for it on the mini rack here.

One thing off the top of my head that you could do would be to angle the Pimoroni display out to keep it in the space of the single rack unit (1U), which would free up space in the rack unit below it.

What are you using the displays and the camera for?

monkeymademe, 2025-01-17:
Not sure what angle you mean? sticking out? I'll be honest its not a great solution and I used PLA which warped with the heat... At the moment it was just running the lcars server on a pi which synced to the bigger display. I am looking for another use tbh its living in the bedroom so its not on at the moment.

The cameras are for my tests with my other project https://github.com/monkeymademe/picamera2-WebUI Jeff has mentioned it before on a video

JaredC01, 2025-01-17:
Sticking out a bit, yes.  Basically think of it angled out at about 45 degrees from the top, which shrinks the overall profile.  The idea came from the way the Voron line of printers mounts the display to save space:

[image]

Gotcha on the use case!

monkeymademe, 2025-01-17:
Ha not bad... This version has the pi connected directly at the back so its a chunky setup. But with ribbon cable I would do something like that I think. I like the design you have... While Deskpi has metal and all I feel its wasting too much space and too restrictive on the orientation of the pi (maybe you want it ass backward?).

Anyways I digress I recently broke my v1 7inch display for the pi I would have liked to have that or well now the v2 model fitted to the front or above the system for system monitoring. Something I can kill the backlight too so it does dark.