JaredC01/LabStack #15: Add M4 Mac Mini 1.5U Mount with 1x LabStack Module

JaredC01, 2025-01-19:
> @GarnetSunset single or dual M4 Mac Mini 10inch rack would be amazing, probably 3D printed to save on cost!
> myelectronics.nl has already showcase it's possible for 19" rack  > [Link](https://www.myelectronics.nl/us/19-inch-125u-rack-mount-for-3x-mac-mini-2024.html)
> 
> 
> 
> For a 3D printed mount, it would end up being closer to 1.5U rather than the 1.25U of that mount, just to make sure there's enough material for the weight of the Mac Mini, and for a 10" rack, you wouldn't be able to get more than one Mac Mini.
> 
> The total opening width on the 10" rack is ~222mm (8.75"), and each M4 Mac Mini is 127mm (https://www.apple.com/mac-mini/specs/).
> 
> Thinking out loud, I could probably make a 1.5U mount for the M4 Mac Mini, plus a single LabStack module on its side, and MAYBE a keystone jack for the MacMini's Ethernet port and/or HDMI port.  I could DEFINITELY make a 2U version, but would still only be able to add a single LabStack module, but it would allow multiple keystone jacks for the Mac Mini...
> 
> Let me know if there's interest, and I'll see about adding it to my list for LabStack modules (so long as Jeff doesn't mind the hijack). 

 _Originally posted by @JaredC01 in [#56](https://github.com/geerlingguy/mini-rack/issues/56#issuecomment-2600975826)_

GarnetSunset, 2025-01-20:
Extremely excited and ill be watching this issue to test it out!

JaredC01, 2025-01-20:
Added to the repo and awaiting feedback.  Info referenced here: https://github.com/JaredC01/LabStack/issues/new/choose

JaredC01, 2025-01-20:
> Extremely excited and ill be watching this issue to test it out! 

Mind reposting your info here?

For the power button; I knew it would be an issue, but I'm not sure of the best way to solve it yet. This will require a compound mechanism of some sort to change axis of motion for the button, but I'll lookin into it.

For the looseness, it looked like there was a pretty decent gap above your Mac Mini in the printed part... can you get me a close-up of that? I can adjust the height for the model, but was using someone else's CAD model.

JaredC01, 2025-01-20:
@GarnetSunset I updated the design a bit for the mount itself.  It's a bit thicker on the actual Mac mounting portion, and I adjusted the fit to be tighter top and bottom.  There's also new spacers that should physically touch the Mac Mini to keep it from moving anywhere.

The only concern I have with the design is that the layer lines for the Mac mount are in the weakest direction, which is why I added a bit more material, but I may look at adding a couple of separate flat sheets in the stronger direction to sandwich everything together.

If you don't mind printing another one, please let me know how the Mac fit is, and if it seems sturdy enough for the weight of the Mac!

GarnetSunset, 2025-01-20:
Wow! Thank you so much!! (Sorry I was out)

I'll give it a shot.... I'll print as draft to make it work nicely

(I'm planning on putting a jetkvm next to the m4 mac mini, and im wondering if the extension boards can do barrel plugs?) 

But maybe a hole to push the power button manually? Just as a "in case of emergency" in the mean time? 

For the record, my old message: 
```
It's IN and it WORKS! 

Two things that should *probably* be addressed perhaps:

1. The power button on the bottom left side is inaccessible, I realize this is apples fault not yours, but it *might* be important to people :) (not blaming you in the slightest!!)
2. Despite the spacers, it's not "snug" enough in there and can spin a lil and fall out!

Otherwise I love it!! Thank you Jared!
```
[image]
[image]

JaredC01, 2025-01-21:
Let me know how it goes this go around.  For the power button, I worry about adding a hole in a part that's already in the weak orientation...  Maybe I'll add a lever that comes out of the front somehow...

GarnetSunset, 2025-01-21:
Fits mUCH better! Much more snug.

JaredC01, 2025-01-21:
Awesome!  Any other feedback before I call it good?

GarnetSunset, 2025-01-22:
I like it! Only thing I would change is the button obviously :) I say its good!!

JaredC01, 2025-01-22:
I'll see what I can do for the button!

GarnetSunset, 2025-01-22:
Ah!! One thing, not *strictly* related, but would be nice, a rotated JetKVM mount for that spot so it's not going to be 90degrees?

GarnetSunset, 2025-01-22:
> [[image]@GarnetSunset](https://github.com/GarnetSunset?rgh-link-date=2025-01-22T20%3A30%3A52.000Z) Which model module did you used for the tp-link?

Dunno! Just grabbed a 10 inch mount for my tp SG108E

JaredC01, 2025-01-31:
> Ah!! One thing, not _strictly_ related, but would be nice, a rotated JetKVM mount for that spot so it's not going to be 90degrees?

Just coming back to this... I'll have one sorted for you shortly!

JaredC01, 2025-01-31:
> Woot!!!

Module added... Single JetKVM and single Keystone jack in a 90-degree module.  @GarnetSunset please let me know how it fits!

JaredC01, 2025-01-31:
> Thank you!!!! :)

You're welcome!  I expect a few pictures of both the updated Mac Mini mount, and the rotated JetKVM Module when it's done!

GarnetSunset, 2025-01-31:
Sir yes sir! I'm waiting for my Jetkvm unit still... But otherwise I'm super down!

customade, 2025-08-04:
Hi; I used this power button [3D print](https://www.printables.com/model/1074723-mac-mini-m4-power-button) and it works pretty well but it blocks off one of the stands.