JaredC01/LabStack #16: Module for Waveshare Mini Base Board (A/B)

shanduur, 2025-01-21:
Hi! First of all, thank you for making such an amazing project!

I came here with a request, for modules that will allow using Raspberry Pi Compute Modules (and other with the same format) in a LabStack using the following base boards / carrier boards. 
 
## CM4 I/O Base (version A)

**URL:** https://www.waveshare.com/wiki/CM4-IO-BASE-A

| Dimensions | Details |
|---|---|
| [image] | [image] |

No. | Component | Description
-- | -- | --
1 | CM4 connector | Suitable for all versions of Compute Module 4
2 | DC power supply/programming interface | 5V/2.5A power supply, also can be used as an eMMC programming interface
3 | DISP Interface | MIPI DSI Display Interface
4 | FAN header | For connecting cooling fan, allows speed adjustment and measurement, only supports 5V fan.
5 | CAM Interface | Dual MIPI CSI camera interface
6 | HDMI0 Interface | HDMI Interface, Support 4K 30fps output
7 | USB 2.0 Interface | 2-channel USB 2.0 Interface, for connecting sorts of USB devices
8 | Gigabit Ethernet | Gigabit Ethernet RJ45 connector, with 10 / 100 / 1000M network support
9 | M.2 indicator | Indicating the operating status of M.2 interface
10 | ACT indicator | Raspberry Pi operating status indicator
11 | PWR indicator | Raspberry Pi power indicator
12 | BOOT selection | Jumper shorted: CM4 would be booted from USB-C interface Jumper opened: CM4 would be booted from eMMC or Micro SD card
13 | 40PIN GPIO header | Conveniently connect various HAT modules
14 | Micro SD Card Slot | For connecting a Micro SD card with a pre-burnt image (Lite variant ONLY).
15 | HDMI1 interface | HDMI1 Interface, Support 4K 30fps output
16 | USB 2.0 connector | Can be connected through an adapter
17 | FE1.1S | USB HUB chip, expanding one USB port to 4x ports
18 | M.2 Interface | Supports sorts of NVME SSD, or communication modules with PCIE M.2 KEY-M interface

## CM4 I/O Base (version B)

**URL:** https://www.waveshare.com/wiki/CM4-IO-BASE-B

The major difference is RTC battery and circuitry on the bottom, visible on 2nd image

| Dimensions | Details |
|---|---|
| [image] | [image] |

No. | Component | Description
-- | -- | --
1 | CM4 connector | Suitable for all versions of Compute Module 4
2 | DC power supply/programming interface | 5V/2.5A power supply, also can be used as an eMMC programming interface
3 | DISP0 interface | MIPI DSI Display Interface, DISP0 interface
4 | FAN header | For connecting cooling fan, allows speed adjustment and measurements, only supports 5V fan.
5 | CAM interface | Dual MIPI CSI camera interface
6 | HDMI0 interface | HDMI Interface, Support 4K 30fps output
7 | USB 2.0 interface | 2-channel USB 2.0 Interface, for connecting sorts of USB devices
8 | Gigabit Ethernet | Gigabit Ethernet RJ45 connector, with 10 / 100 / 1000M network support
9 | M.2 indicators | Indicating the operating status of the M.2 interface
10 | ACT indicators | Raspberry Pi operating status indicator
11 | PWR indicators | Raspberry Pi power indicator
12 | BOOT selection | jumper connected (ON): CM4 would be booted from USB-C interfacejumper not connected (OFF): CM4 would be booted from eMMC or Micro SD card
13 | 40PIN GPIO header | Conveniently connect various HAT modules
14 | Micro SD card slot | For connecting a Micro SD card with a pre-burnt image (Lite variant ONLY).
15 | HDMI1 port | HDMI1 port, Support 4K 30fps output
16 | USB 2.0 interface | Can be connected through an adapter
17 | FE1.1S | USB HUB chip, expanding one USB port to 4x ports
18 | M.2 connector | Supports sorts of NVME SSD or communication modules with PCI-E M.2 KEY-M interface
19 | RTC | PCF85063 chip, Allows RTC-related functions like wakeup, shutdown, reboot, and more
20 | RTC interrupt pin switch | PI-RUN: CM4 reboot on RTC interruptionGN-EN: CM4 shutdown on RTC interruptionD4: D4 pin is triggered on RTC interruption
21 | EMC2301 | Fan controller, for speed adjustment/measurement
22 | RTC battery socket | Can be connected to CR1220 button battery

JaredC01, 2025-01-22:
@geerlingguy do you have any of these I can use for measurements?

geerlingguy, 2025-01-22:
I have one sitting on the desk in front of me — note that the hole locations and sizing matches the Pi 4/Pi 5 practically the same, it just needs a little extra bottom height for the extra goodies there.

geerlingguy, 2025-01-22:
5mm seems like it would work fine, though you could go to 6-7mm for a little extra clearance (to keep any hot NVMe SSDs from heating the printed surface too much).

Alternatively, you could do cutouts or at least indents into the printed part for the NVMe or other taller components.

JaredC01, 2025-01-22:
> 5mm seems like it would work fine, though you could go to 6-7mm for a little extra clearance (to keep any hot NVMe SSDs from heating the printed surface too much).
> 
> Alternatively, you could do cutouts or at least indents into the printed part for the NVMe or other taller components.

Thanks! I'll see what a cutout looks like behind it, and whether or not I can do a dual-mount version!

JaredC01, 2025-01-24:
I've added a module for the WaveShare CM4-IO-BASE carrier(s), but it needs testing.  @shanduur and @geerlingguy if one of you wants to print the mount and let me know if their CAD model is accurate, namely on the front cutout fitment, I would appreciate it!  From their CAD model, it looks like the ports will be recessed a bit more than normal, but there's also ribbon connectors on the bottom, so I don't want to move the board forward any more than is necessary.  The big concern is the HDMI port and seating fully with a cable plugged in.

Just for reference, I ended up making the spacing 7mm per @geerlingguy's note just to give some breathing room, and I also added a cutout on the back to access the M.2 drive with the carrier installed.  There's 2.85mm of space between the tallest component on the rear (the battery holder) and the base of the mount, so I expect there won't be any issues with fitment!