JaredC01/LabStack #18: Request for Intel Nuc BOXNUC8i5BEK2

bemany, 2025-01-22:
> Request for Intel Nuc BOXNUC8i5BEK2
> 
> I am not able to find a simple solution for mounting the nuc on its mounting screws on a 10" module
> 
> [image]
> https://www.intel.com/content/www/us/en/support/articles/000006857/intel-nuc/intel-nuc-kits.html
> 
> It would also be nice to ziptie the power adapter next to it "huntkey HKA09019047-6u"
> https://www.ebay.de/itm/256636100141
> 
> Break out of HDMI, Keystone .. is not mandatory 

 _Originally posted by @bemany in [#10](https://github.com/JaredC01/LabStack/issues/10#issuecomment-2604327927)_

CodyKochmann, 2025-01-25:
+1 for a nuc8 option! It would be amazing to get those puppies vertical. The 4 screws along the bottom might be a better target to design a mount for than the provided nuc mount that makes the nucs footprint bigger.

JaredC01, 2025-01-31:
This will end up being part of the LabStack Rack repo - will cross-post there once ready.