JaredC01/LabStack #19: Module request. RPI5 + base nvme

fmurodov, 2025-01-24:
First thank you for this project. 

I have Pi5 with Pimoroni nvme base. Is it any change to make module for this? 

https://shop.pimoroni.com/products/nvme-base

JaredC01, 2025-01-24:
Module has been added to the Modules folder.  I used Pimoroni's CAD model, and their standoff height listed in the product page (7mm) for the spacing.  You will need to put screws thru the back of the mount into the 7mm spacers, then screw the Pi in from the top to the spacers.

Let me know if it fits, and if there's any questions or issues!

fmurodov, 2025-01-25:
I think the front panel needs to be flipped 180 degrees because the recesses for the NVMe board are currently on top of the PI instead of at the bottom.

[image]

JaredC01, 2025-01-25:
> I think the front panel needs to be flipped 180 degrees because the recesses for the NVMe board are currently on top of the PI instead of at the bottom.
> 
> [image]

Ah! I forgot to check clearance there! I'm not used to bases yet. 😅

The above is for adding a HAT still,  but if the Pimoroni base needs a bit of extra, I'll extend it. Give me about 30 minutes and I'll have a new version for you!

JaredC01, 2025-01-25:
Updated version pushed to the repo.  Let me know how this one goes, but it should fit just fine!

fmurodov, 2025-01-25:
Perfect. Thanks a lot!

[image]

[image]

closed 2025-01-25 by fmurodov