JaredC01/LabStack #21: Module request: NanoKVM lite

japz, 2025-01-26:
Thank you for starting this project, it's amazing!

I suggest a new module for the [NanoKVM lite](https://github.com/sipeed/NanoKVM?tab=readme-ov-file
). It's very small, two will certainly fit in a slot, and it might even be possible to squeeze three or four in. It would pair nicely with the Radxa X4 - the USB and HDMI can be routed behind the panel and the ethernet port could face out.

It is basically a LicheeRV board and there are some drawings and 3d files of the Lichee Nano board available via: https://cn.dl.sipeed.com/shareURL/LICHEE/LicheeRV_Nano.
I think there would need to be a small cutout on the bottom for the heatsink of the chip.

JaredC01, 2025-01-28:
I'll see what I can do.  The CAD for the LicheeRV doesn't include the HDMI port, so I would ideally need some measurements from the NanoKVM directly, at least with spacing from the bottom of the HDMI port to the top of the PCB or USB-C port.

japz, 2025-01-29:
Thanks! Here you go:
[image]

Let me know if you need any additional measurements.