JaredC01/LabStack #22: Ubiquity Toolless Rack Mount

JOLLY6R33NGIANT, 2025-01-27:
A mount for the ubiquity rack with room for a couple jetkvms, Rpi5s and some keystone cutouts I think would be nice.

JaredC01, 2025-01-27:
Just so I'm clear, do you mean the toolless 19" Ubiquiti rack? https://store.ui.com/us/en/products/toolless-mini-rack

With the way their rack system works, there are no rack ears, which means mounting would be basically impossible with a traditional design.  While I won't rule it out, I don't think this will end up being supported.

JOLLY6R33NGIANT, 2025-01-28:
Yeah that’s the one. I saw someone in the jet discord post about something similar so I thought it was worth a shot bringing up here.

[image]