JaredC01/LabStack #23: Fusion 360 Share link or STEP Files

nlj007, 2025-01-30:
This may exist somewhere, but I can't find it. Would it be possible to share this CAD data via a Fusion link as shown [in their documentation](https://help.autodesk.com/view/fusion360/ENU/?guid=ASM-SHARE-DESIGN)? 

I have not been able to open the .f3d file in Fusion, and all resources online point to sharing a link to be the best way.

If sharing the native file this way is not an option, a non closed-source filetype such as parasolid (.xt) or STEP would be much appreciated. STL's are much less friendly to work with in parametric modelling software (SolidWords, OnShape, Fusion).

For reference, I am hoping to import most of these designs to OnShape and share them publicly for others to access and modify.

JaredC01, 2025-01-30:
I have no plans to share via link because I'm doing a lot of prototyping and addition of new models, and the link would point to the latest save, so there's a very real chance that people may download half-finished models.  The full F3D file shouldn't have any issues opening in Fusion, and I just tested downloading and opening it myself, which worked fine.  The file is rather large, so if you're having trouble loading it, it's likely due to the components included (namely the Pi models since they have a ton of components).  Give it a bit of time if your machine isn't super powerful, but it should load.

I had originally exported a STEP file, but the file size was too large to upload to GitHub without pushing it to a large-file repository, and I'm currently working on separating out the LabStack modules and mounts from the rackmount items that are compatible with LabStack (which I'm calling LabStack Rack) to make the CAD files smaller.

Once the repo's have been updated with everything, I'll have STEP files for everything, but in the meantime, I would try the Fusion file again, giving your machine time to load it.

JaredC01, 2025-01-31:
CAD updated, including simplifying all of the Pi and board models to save space.  STEP file was still too large to upload directly, but being text-based, I was able to compress it into a ZIP file.  Download and unzip for the full STEP.

closed 2025-01-31 by JaredC01