JaredC01/LabStack #27: 120mm fan Panel

laplansk, 2025-02-06:
Jared,

I see that there is a panel available for 80mm fans. Would you be able/willing to adapt to create a similar one for 120mm fans? I have several Noctuas laying around that I'd like to put to use rather than purchasing new 80mm fans. I also think that it will be nice to get air flowing full-width of the rack with two side-by-side 120s. They are admittedly just a tad too wide for the rack when side-by-side, but it looks to me like this could be accounted for by either including standoffs in the model so that they can be mounted on the inside of the case OR just mounting them on the outside of the panel. (If that wasn't clear, I can try to draw what I am describing and attach it to this Issue or we can have a quick chat about it). This whole thing is an excellent project. I am in the middle of printing the drawer now and really excited to get my homelab set up all tidy. 

Cheers and thank you for considering. 

[image]

JaredC01, 2025-02-06:
I'll take a look in CAD and see if anything makes sense... 2x 120's certainly won't fit inside the rack, but I may be able to do something clever to make them work as the picture shows externally.

laplansk, 2025-02-06:
Appreciate you looking into it.

For context, this is what I was imagining when I said inside. [image]

Looking at how the screw holes on the fans would line up, I suppose you'd probably have to mount the panel on the inside to make it work, but it _could_ work, right?

JaredC01, 2025-02-06:
Without access to the mount holes on the fans, it would be hard to install that way. Off the top of my head, the best solution to make the internal idea work would be to make a plate that the fans mount on and sits internally, then make an external plate that can interface with the internal plate... I'm not sure whether that will work though, as the screws for the front panel would interfere with the fans.

I should have some time later tonight or tomorrow night to take a closer look at this and see what I can decide on.

JaredC01, 2025-02-09:
Just keeping this up to date... I haven't had time to poke at this at all.  I'll update you when I get some time!

laplansk, 2025-02-18:
@JaredC01, 

I am new to CAD (basically 0 experience), but wondering if there is a intro/primer course you might want to point me to in case I can ramp up quickly and take a shot at modifying the existing 80mm fan chassis/[anel myself. Let me know! This is a skill I would like to acquire and I'd love to contribute!

JaredC01, 2025-02-18:
I'm sure there are plenty of good references available, but I learned most of what I know just by googling what I was trying to do.  Send me a DM on Discord if you have it, I'll see if I can help you out any...  I do still have this on my radar, but have been swamped with work from starting a new position.