JaredC01/LabStack #28: Organized list for which files correspond to which project?

xksteven, 2025-02-07:
Because it's my first time building one of these and I consider myself a noob I worry that I'll be picking the wrong file or stl file when sending it off to be made.  To be clear I'm considering this frame: [GeeekPi / DeskPi RackMate T0](https://amzn.to/4iIBsZ6)

It'd be nice to have a list (for example for your build).  I used these 4-5 files.

I don't know which of these two to get?
* [20M3x4x5mm](https://github.com/JaredC01/LabStack/blob/main/LabStack%20Mini/STLs/2U%20Mini%204x%20LabStack%20Module%20M3x4x5mm%20Heatset%20Bracket.stl)
OR
* [M3x3x4.6mm](https://github.com/JaredC01/LabStack/blob/main/LabStack%20Mini/STLs/2U%20Mini%204x%20LabStack%20Module%20M3x3x4.6mm%20Heatset%20Bracket.stl)

I think the JetKVM file is another directory?

* [JetKVM](https://github.com/JaredC01/LabStack/blob/main/Modules/STLs/Module%201x%20JetKVM.stl) (With a few variants for orientation or 2.)

JaredC01, 2025-02-09:
It is all a bit of a mess at the moment... I've been working on adding modules rather than getting the documentation done (with my thought being that I would make some documentation for each module).

For the printed bracket, the "normal" heatset inserts I've suggested (info over on the LabStack Rack repo, formerly JetKVM repo) are M3x4x5 (M3 thread, 4mm deep, 5mm wide).  I recently merged a PR that added the CNC Kitchen "shallow" heatsets that are M3x3x4.6 (M3 thread, 3mm deep, 4.6mm wide).  Chances are you will want the M3x4x5 version.

All of the modules are in their own directory as they can be used with both LabStack Mini (the 10" version) and LabStack 19 (the full rack version).  The separation between the modules and brackets will remain after cleanup, just for recorrd.

axiopaladin, 2025-02-10:
Personally I think some initial documentation would be the single biggest upgrade to this project right now. It doesn't have to go into full detail with specialized assembly instruction and everything; just a BOM and one-sentence description for each model would be a great help to getting started. Are there any docs or notes anywhere that we could reference to figure that out? I can try putting a PR together with a basic structure for things if that would help.

JaredC01, 2025-02-11:
I 100% agree with you (and it's well past due!).  I should have some time tomorrow or Wednesday to at least get a rough initial set of instructions set up.  From there, I'm open to updates as needed!

axiopaladin, 2025-03-01:
Are the any updates to share on the timeline for this?

itsCARLsanity, 2025-03-07:
@axiopaladin see Issue #42