JaredC01/LabStack #29: Radxa x4 mounts

wmblue, 2025-02-09:
I can't tell how you're mounting the Radxa x4 to the 3d printed plate mounts?

can you show me how you're mounting it with pictures, as well as what hardware/screws you might have needed to mount it?

Thanks!

JaredC01, 2025-02-09:
You need one set of spacers (either the PoE HAT spacers, or the longer ones if not using the PoE HAT), then it goes together backwards (screws through the back of the mount).  The screws are M2.5x25mm (head type doesn't matter)

[image]

[image]

wmblue, 2025-02-09:
Awesome! ty!