JaredC01/LabStack #33: LabStack/Modules/STLs /Module 1x Pi with 25 SSD

OregonGunslinger, 2025-02-13:
Would it be possible to add pass-thru holes in the bottom plate so that screws can be installed from underneath to attach the pi?

JaredC01, 2025-02-13:
Pretty sure I understand what you mean, but just to clarify, you need access to the Pi's mount holes through the SSD's plate, correct?

https://github.com/JaredC01/LabStack/blob/main/Modules/STLs/Module%201x%20Pi%20with%2025%20SSD.stl

OregonGunslinger, 2025-02-13:
Exactly.  Thank You.

JaredC01, 2025-02-13:
Added 4mm holes through the SSD bracket and updated the original file with changes (updated in my last link).  If that's it, I'll close the ticket!

OregonGunslinger, 2025-02-13:
That's perfect for this one.  I have another ask (different situation) but I can open a new issue.

closed 2025-02-13 by JaredC01

OregonGunslinger, 2025-02-22:
Follow up, if the new holes could be 6mm, that would help fit screws with slightly larger heads. The screws I'm using are 5.54mm.