JaredC01/LabStack #34: Module 1x Pi with POE and SSD

OregonGunslinger, 2025-02-13:
I have a Pi4 with:

GeeekPi Isolated PoE HAT for Raspberry Pi 4
https://www.amazon.com/dp/B08CVDQWXF
-and-
GeeekPi M.2 NVME SSD Storage Expansion Board for Raspberry Pi 4
https://www.amazon.com/dp/B0CBJYWKJ1

[image]

[image]

[image]

[image]

Would it be possible to create a tray for this?

JaredC01, 2025-02-13:
I'll see what I can do.  Can you get me the standoff size on both the NVMe base to the bottom, and the NVMe base to the Pi?  They look to be the same size, but I'd like to verify.

OregonGunslinger, 2025-02-13:
Without getting the calipers, I'm showing that both are 1/4 inch standoffs.  Note that in the picture, the mount for the NVMe is wrong and should be a bit shorter  but that shouldn't affect the part.

JaredC01, 2025-02-13:
> Without getting the calipers, I'm showing that both are 1/4 inch standoffs. Note that in the picture, the mount for the NVMe is wrong and should be a bit shorter but that shouldn't affect the part.

They're usually marked in mm rather than inches, which means they're either 6mm or 7mm (with 1/4" being closer to 6mm).  If you get a chance to verify that, please do.  I've seen 7mm standoffs pretty commonly, but not often for 6mm (and being 1mm off means the ports in the plastic won't line up).

I'm not sure what you mean by the mount for the NVMe is wrong; are you referring to the standoffs on that side?

OregonGunslinger, 2025-02-13:
Yes, the standoff.  In the image, the NVMe drive is angled funny and shouldn't be.  I'll get the calipers and supply a precise measurement for the other standoffs.

OregonGunslinger, 2025-02-13:
Sorry to muddy the waters.  The NVMe standoff is too long so the drive is offset too much on that end.  I do have a shorter standoff that will true it up.

The pi standoffs are 7mm by the micrometer.

JaredC01, 2025-02-13:
Good to know on the Pi Standoffs!  For the standoff from the NVMe base, I can model my own "standoffs" with the printed part.

If 7mm works with your current setup, I might as well leave the 7mm spacing (it's also what's currently on the WaveShare CM4 IO board, so an easy transfer).

I don't see any sort of CAD for the board design, so I will need to do some testing on spacing between the USB port on the base and the USB port on the Pi...  Are you able to get a measurement for me between the USB-A to USB-A ports on both the insides and the outsides?  Basically the inner part of the U and the outer part of the U...  That will get me the dimensions for where the USB port needs to be in the cutout!

OregonGunslinger, 2025-02-13:
Pardon my lack of drawing skills...  I hope this helps:

[image]

JaredC01, 2025-02-13:
That's perfect, thank you!

DerAndereAndi, 2025-03-14:
I'd like to share some thoughts on getting such a setup work with small hacky modifications of a current module.

I added a negative part to extend the opening of the "Module 1x Pi with HAT" vertically to the top and use the distance elements (same as in the images) to mount them. One problem turned out to be to bottom extension board for the SSD being a bit longer to the front side compared to the Raspberry Pi board. And secondly my "PoE HAT (C)" from waveshare is a bit taller so I can't slide it in easily. I have to detach the PoE HAT for now and reattach it. So getting the whole setup more closely to the bottom would be helpful in general. E.g. by not having to use the distance pieces at the bottom and there safe some vertical space and the NVME still have some space "to breath" :)