JaredC01/LabStack #35: Bracket Module Suggestion for the Radxa Penta SATA Hat (Raspberry Pi 5)

Ash-Bash, 2025-02-13:
I have a Radxa Penta SATA Hat for Raspberry Pi 5, and would like the ability to Mount it inside The DeskPi Rackmate T0, would be useful for anyone who's making a NAS Server Rack. Ive provide a Image of what the Hat looks like on the Raspberry Pi and a Link to their official page for the hat (For the Specs etc).

[Radix Penta SATA Hat Official Page](https://radxa.com/products/accessories/penta-sata-hat)

[image]

JaredC01, 2025-02-13:
What's the expectation for integration here?  Are you thinking about a LabStack module, or just some sort of option for mounting it with 4 drives in a mini rack?  I assume you would be looking for something OTHER than a LabStack module, mainly because there's not enough room for the drives to mount, but figured I would verify the intent!

Ash-Bash, 2025-02-13:
> What's the expectation for integration here? Are you thinking about a LabStack module, or just some sort of option for mounting it with 4 drives in a mini rack? I assume you would be looking for [...]

To make a mini NAS server, Describing it to the best of my ability, the Hard Drives would be on a separate shelf/level from the raspberry Pi on the rack, they'd be laying flat in their own hard drive trays plan is for 5 hard drives in 2 stacks, but I am using the Hard Drives DeskPi's Official Solution for the Mounting of the Hard Drive Trays for the time being.

Ash-Bash, 2025-02-13:
> Okay, so my next question would be, how are you breaking out the SATA from the HAT to another shelf? I'm trying to get a full layout in my head of how this is going together so that I understand whether this would work as a LabStack Module or not.

The plan is to use SATA Data+Power Combo Extension Cables To & From the Hat & Hard Drives, I guess the Raspberry Pi could go Sideways like other Modules, the cables wouldn't need to come out of the front, since the cabling would be routed to the back of the hard drives since the connector would inwards so to speak, the ESATA port is the outlier here, where it would need to be threaded back into the rack they do provide a ESata to SATA Data+Power Cable. 

HAT has a few Power Options A ATX Power Supply Interface and a Power Barrel Jack, but that would be a Internal too.

JaredC01, 2025-02-13:
Okay, I'm on the same page now!  The cabling for the drives will probably still be a bit too tall for a module, but we can try going that route and see what happens.  This is a similar setup to what I had here:

[image]

I eventually moved to slim cables, but still ended up being too tall (you had to install the module into the LabStack bracket before installing the bracket).

I'll see what I can do.

HalmSascha, 2025-02-14:
I thought about almost the identical use case and maybe there would be a solution for it. If there would be a bracket like the 2U Mini 4x LabStack Module M3x3x4.6mm Heatset Bracket with only one hole for a LabStack Module on the right side, the RasPi5 with the Radxa Penta SATA Hat could be mounted horizontally with a new/customized LabStack Module (Module 1x Pi with 25 SSD where the eSATA connection can be used. So there would be enough space for the 4 mounted SSDs horizontally and an additional 2.5 SSD connected with the eSATA-Port.

Ash-Bash, 2025-02-14:
> I thought about almost the identical use case and maybe there would be a solution for it. If there would be a bracket like the 2U Mini 4x LabStack Module M3x3x4.6mm Heatset Bracket with only one hole for a LabStack Module on the right side, the RasPi5 with the Radxa Penta SATA Hat could be mounted horizontally with a new/customized LabStack Module (Module 1x Pi with 25 SSD where the eSATA connection can be used. So there would be enough space for the 4 mounted SSDs horizontally and an additional 2.5 SSD connected with the eSATA-Port.
> 
> What I don't know at the moment is whether the 4 horizontally mounted SSDs could be too heavy for the LabStack module.

Yeah, that’s what my concerns would be for packaging reasons I decided that the hard drives will be on another shelf, and use the lab stack shelf for other raspberry pis / kvms. Since I’m interested in also have docker images running too, to run other auxiliaries systems, like personal git server, homebridge etc

JaredC01, 2025-02-14:
Looking at this with a set of fresh eyes this morning, I have a couple of additional challenges that make me wonder if it's not worth using another solution (like the NVMe to SATA adapter setup I posted above)...

The hardest things to work around / problems that need solving that I can see so far are as follows...
* For a LabStack Module, the SATA cables will likely be too tall, meaning the slot next to the module would be unused with a blank cover, and you'll likely have to assemble the LabStack bracket and module together before putting it into the rack.  This isn't a dealbreaker, but it makes reusing that slot(s) hard.
* The Molex connection for power likely won't be usable as it's too close to the front of the Pi and would likely also end up being too tall.  So long as you're using the barrel jack instead, this should be okay.
* The eSATA connector will absolutely have to come out of the front of the panel before going to the drive (not necessarily an issue, but not "clean".

Here was my Pi 5 + PoE/NVMe HAT with M.2 to SATA adapter installed in a LabStack Mini...  You can see it "works" but it cuts into the module next to it.

[image]

What I ended up doing with my rack was setting up an M.2 extender to put the M.2 to SATA adapter back further in the rack so that all of my LabStack Mini slots were usable.

[image]

I don't have an issue continuing on this if that's what you would like, but I think there's enough caveats for this specific HAT for this specific use-case that it would likely be better used in a standalone NAS setup rather than in the LabStack / Mini Rack ecosystem...

Let me know if you want to continue with this specific setup, or if you have any questions about alternative options...

HalmSascha, 2025-02-16:
From my point of view and looking at my use case I'd go with a standalone NAS. But thank for sharing your thoughts on this.

Welved, 2025-05-07:
Question on that RaspberryPi 8 bay 2U. Is each of those bays able to accomidate a hackergadget PoE + NVME hat (https://hackergadgets.com/products/nvme-and-poe-hat-for-raspberry-pi-5)? Also, where would I go to locate that 8 bay 2U Pi solution?

JaredC01, 2025-05-08:
> Question on that RaspberryPi 8 bay 2U. Is each of those bays able to accomidate a hackergadget PoE + NVME hat (https://hackergadgets.com/products/nvme-and-poe-hat-for-raspberry-pi-5)? Also, where would I go to locate that 8 bay 2U Pi solution?

Are you looking for a mini rack or full-size rack?

For the Pi's, all of the mounts should be able to accommodate those HATs from looking at the visuals, but it would be worth verifying with one before going whole ham (as they say) with 8 of them.

Welved, 2025-05-08:
> Are you looking for a mini rack or full-size rack?
>
> For the Pi's, all of the mounts should be able to accommodate those HATs from looking at the visuals, but it would be worth verifying with one [...]

I already have a deskpi 10 inch 8U, I was just wondering if a bay could accomidate a hackgadget PoE + NVME hat, as that is what I have.