JaredC01/LabStack #36: Fan setup

alikaragoz, 2025-02-14:
Hey Jared, I was curious about your fans setup. Can you share some info about the model you used and how you wired them to the pi? 🙏🏻

JaredC01, 2025-02-14:
Absolutely!  For the fans, I used a set of Arctic Cooling 80mm fans: https://www.amazon.com/dp/B07XL9QJHF

The fans themselves are powered by the PoE + NVMe HAT that I have on my Pi 5, which has headers for 3v3, 5v, and 12v (Dupont connectors), but you can grab 5v power from just about anywhere to run them.  They work very well at 5v, and are nearly silent while pushing/pulling what I consider to be a "lot" of air.

Here's the PoE + NVMe HAT I used: https://www.amazon.com/dp/B0D8JC3MXQ

You can see the pin headers in the corner, but you can also just use the 5v and GND pints from the 40-pin on the Pi if you aren't using a HAT that blocks them!

alikaragoz, 2025-02-14:
I’ll take a look at the Arctic fans; they’re also quite affordable in France.
I’d like to find a way to power both fans using a single USB port—I think a USB splitter could work.
I really appreciate all the time and resources you’re providing on this repo. Thank you 😇

closed 2025-02-14 by alikaragoz