JaredC01/LabStack #39: Heat set inserts in module trays

ReillyB, 2025-02-19:
Quick question since I couldn't find pictures of the trays outside of the mounting plate. It looks like the modules have holes for heat set inserts but I am unclear as to how they are supposed to be used. Could a couple pictures of the modules from different angles be added to the ReadMe to illustrate how those heat sets are intended to be used? Maybe I'm just dense and missing the obvious use case lol.

ReillyB, 2025-02-19:
[image]

ReillyB, 2025-02-19:
AH, they're for the Pi5 active cooler pins aren't they? Welp, I'll close the issue since they're not for heatsets at all!

closed 2025-02-19 by ReillyB

JaredC01, 2025-02-19:
> AH, they're for the Pi5 active cooler pins aren't they? Welp, I'll close the issue since they're not for heatsets at all!

Sorry for the late reply; you are absolutely correct!