JaredC01/LabStack #41: Sell and shipping

cmonty14, 2025-02-24:
This is great work!
@JaredC01 Do you sell any of the modules you designed?

halkeye, 2025-02-25:
https://github.com/JaredC01/LabStack/issues/24#issuecomment-2624656200
he already said he doesn't have interest in selling / business them

helmarw, 2025-02-25:
@cmonty14 if you are located in Europe i could print them for you in SLS for a fee

helmarw, 2025-02-25:
> > [@cmonty14](https://github.com/cmonty14) if you are located in Europe i could print them for you in SLS for a fee
> 
> I'm located in Germany. Could you share the costs for a module + shipping? This is just a starting point, I certainly need more than 1 module.

na das ist ja noch einfacher ;)
schick mir am besten ne mail, müsste in meinem Profil sein