JaredC01/LabStack #58: Pimoroni NVME base with RP5

mmbcode, 2025-07-16:
New to 3D printing, and running into a bit of an issue with the pimoroni NVME base RP5 STL. My selected NVME's have large heat sinks requiring me to use 12MM offsets in place of the default 7MM that come with the base. Basically I need to move the IO shield rectangle up 5MM from its current location (in reference to the original base of the where the mounting of the device would occur. The DeskPI rack comes with a metal 2U mounting bracket that use centered thumb screw  (32-6) rather than the two holes on top and bottom per module as included in you STL. 

Question part: I have been trying to make these changes to your posted STL and failing miserable. Rather than make a bunch of wrong turns is there a software you would recommend to make these changes? I was hoping it would be as simple as selecting the negative space and moving it around, but the programs I was attempting to do this tinkercad can't seem to ungroup things. Can you recommend a good software to make these changes to the STLs?

Thank you in advance for your time!

MMB

JaredC01, 2025-07-16:
I can make that happen... give me 24 hours and I'll have an option for you.

JaredC01, 2025-07-17:
Well, make that 48 hours, but we're good to go!  Added a "tall" version of the mount, to be used with 12mm standoffs rather than 7mm.

https://github.com/JaredC01/LabStack/blob/main/LabStack%20Modules/STLs/Module%201x%20Pi%205%20with%20Pimoroni%20NVMe%20Base%20Tall.stl

closed 2025-07-17 by JaredC01