JaredC01/LabStack #59: 2x Pi module - Base/HAT support question

dgmulholland, 2025-07-17:
Hey, awesome project!

I'm looking at building a high-density pi cluster and was hoping you could give an insight into the amount of space available in the 2xPi module for HATs or Bases? I want to put 8 Pi 5's in 2U using 4 of the the 2x Pi modules but need a little extra space for POE and storage.

My preferred solution would be the [Waveshare PoE M.2 HAT+ (B)](https://thepihut.com/products/poe-m-2-hat-b-for-raspberry-pi-5) as it supports full size NVMe and PCIe 3.0 speeds but it also sits on top of the front I/O and increases the total height. Looking at the STL am I right in assuming that anything outwith the volume of the Pi itself will cause issues?

JaredC01, 2025-07-17:
More or less correct... You have about 3mm of space above each Pi before you run into issues, and that HAT will absolutely take more space than that.  There's also the issue of mounting for both Pi's since they share a set of mounting holes, and the HATs use standoffs....  You would need to come up with something clever (like using a long M2.5 screw and cutting the head off of it?) to make it work there as well.

All that said, there's two options that @geerlingguy reviewed (https://www.youtube.com/watch?v=x9ceI0_r_Kg) that MIGHT work?  The HackerGadgets PoE + NVMe HAT, and the n|fuse PoE HAT with m.2:

https://hackergadgets.com/products/nvme-and-poe-hat-for-raspberry-pi-5

https://n-fuse.co/devices/SBCPoE-RPi-Power-over-Ethernet-Hat-for-Raspberry-Pi-5-with-mPCIe-Slot-m2-slot.html

The limitation there is that you have to use 2230 or 2242 NVMe drives, though 2230 drives in particular seem to be pretty readily available due to the increase in the handheld PC market.

JaredC01, 2025-07-21:
Standoffs with different lengths would work, but not everyone will want to spend extra on that, so I didn't want to assume you would do it the "right way" if you will.

Let me know if you get it working like you want - I'd love to see a cluster of Pi 5's with NVMe HATs and PoE.

dgmulholland, 2025-07-25:
I'm now wondering if the Radxa X4 SBC would be a good candidate for this thought experiment?
It's the same general footprint as the RPi, has a built-in NVMe connector and it's PoE HAT fits within the total volume of the device.

It drinks more power that the Pi that's for sure, and I don't know what the thermals would like considering the chip is on the base and won't have any sort of heatsink other than plastic and another N100 right next to it. I can see it thermal throttling badly.

Possibly a combination of the 1x Pi with HAT and the 2x Pi modules would work to give the required space but reduce the node count to 6? Have the 2x Pi modules on the left and right, then 2 1xPi + HAT modules back-to-back in the middle 2 slots.

JaredC01, 2025-07-25:
I have a Radxa X4 in my mini rack - with the heatsink it takes up a lot of space, and the heatsink is absolutely needed.  I think a Pi 5 with one of the hats above will probably be the best bet, and the n-fuse model is probably the best option of those two...  It's a unique situation - maybe a CM5 with a breakout board?

Radxa for reference:
[image]

dgmulholland, 2025-07-25:
That's a very interesting point... I think I'd written off a bunch of options because I wanted to use full length SSDs but that's not a requirement any more.

I found this from Waveshare which seems to be compatible with the official cooler:
https://thepihut.com/products/poe-hat-h-for-raspberry-pi-5-and-cm5

JaredC01, 2025-07-25:
Actually, the H model doesn't look like it will work - the transformer sticks down below the board quite a bit, and would interfere with the heatsink unless you pull off the fan:

[image]

dgmulholland, 2025-07-25:
I think I've looked at too many HAT's and my brain is a bit fried for a Friday afternoon but would the Waveshare PoE M.2 HAT+ work? Potentially with the tops of the GPIO pins clipped off to reduce the total height?

https://thepihut.com/products/poe-m-2-hat-for-raspberry-pi-5

JaredC01, 2025-07-25:
I don't THINK that one would work, even with the pins clipped, but this one might?  I don't think you can use the fan heatsink with either and keep the height needed, but you might be able to do the thinner heatsink that's out there and squish this guy on top of it...

https://www.waveshare.com/poe-m.2-hat-plus-c.htm

JaredC01, 2025-07-25:
The heatsink...

https://www.waveshare.com/pi5-active-cooler-c.htm

JaredC01, 2025-07-25:
I think I may order a pair of those HATs, heatsinks, and a couple of Pi 5's to test this myself...  It'll be a bit (going to order it from Waveshare directly to save on cost), but I'll let you know how it goes.  If you end up pulling the trigger and get the parts in before I do, let me know how it goes!

dgmulholland, 2025-07-26:
The only place I can see that version of the HAT in the UK is on Amazon (eugh). But I decided to just go ahead and grab it out of curiosity so I'll let you know how it goes and see if I can get some measurements for the total height of the stack

JaredC01, 2025-07-26:
Sounds good. I did end up ordering a pair myself from Waveshare directly, so I'll be testing at well once they arrive.

JaredC01, 2025-07-31:
Oh that's absolutely perfect!  It looks like the heatsink is installed under there as well?  Definitely a compact package - I'm waiting for mine to arrive, and I'll be able to test it in a single 2-Pi module if you can't get a module printed in the meantime, but I think it's going to be absolutely spot-on.

You may want to print one of the 2x 80mm fan 2U brackets and throw a couple of fans in it to cool the Pi 5's since there's not a lot of space for convective cooling.  I love the Arctic Cooling P8's for it - https://www.amazon.com/dp/B07XL9QJHF

Here's the bracket itself - https://github.com/JaredC01/LabStack/blob/main/LabStack%20Mini/Accessories/STLs/2U%20Mini%202x%2080mm%20Fan%20Panel.stl

JaredC01, 2025-07-31:
You're welcome! I'd love to see some pictures when you get it all together!

JaredC01, 2025-08-03:
Well, mine showed up today!  I've only got 1x Pi 5 at the moment (grabbing a couple more from Microcenter tomorrow), but I can say with 100% certainty that these will work in the 2x Pi mount.

[image]

[image]

geerlingguy, 2025-08-03:
Sheesh! How is cooling in there? Seems like a potential winner, as long as you print out of something that won't sag over time...

JaredC01, 2025-08-03:
You know, I may just put together a "vertical" 2x 80mm fan setup that fits in a 1U slot to blow up air from underneath... it may help with something like this, especially if the back of the rack is too far away to get good airflow (like a standard 19" rack).

JaredC01, 2025-08-03:
I may have gone one further...  Dual 4010 Blower fans, and single 4010 Axial fan options... Not sure yet where to get power from since the GPIO is a bit occupied, but that's a later me's problem.

[image]

[image]

dgmulholland, 2025-08-04:
Would a case fan fit on that mount point? I was thinking something like https://noctua.at/en/products/fan/nf-a4x20-pwm. The Noctua fans come with a 30cm extension cable which looks like it might be able to squash under the HAT.

JaredC01, 2025-08-04:
Alright, got the blowers in today - Looks like the USB-C PCB method works, but it is a bit tall.  You can still install it as it sits, but you have to rotate the module to a 45 degree angle to get the PCB through.

I'm reprinting a version with less gap between the Pi and the fans, and one that doesn't use heatset inserts, and I'll post a couple of pics when that's completed.  Once I'm happy with it, I'll release the files as well.

[image]

dgmulholland, 2025-08-07:
So I picked up a couple of fans specifically for the raspberry pi to try and get a plug-and-play option. The official cooler is too tall to have under the PoE hat (as expected), the regular 40mm fan and 30mm blower from the official cooler have cables that don't reach if they are placed at the back. But the 30mm blower marketed as a [replacement for the fan in the Argon Neo 5 case](https://thepihut.com/products/replacement-30mm-fan-for-argon-neo-5-case) looks like a reasonable option assuming it can move enough air to be effective.

[image]

JaredC01, 2025-08-07:
Interesting! Thank you for the research on this as well... the blower MAY be enough, but I will need to create a bracket for it.

Do you have a set of calipers by chance? I need to know how tall the fan is, what the opening size is for the fan output, hole sizes and locations, etc...

I may be able to find a generic 30mm blower model, but I'll have to look!

JaredC01, 2025-08-11:
Next update... The 3007 blower fans I ordered have arrived, but the cables are too short.  I wasn't able to find any other options that had a Pi 5 fan connector pre-installed on Amazon, but I can at least get a model made for the 3007 blowers for those that have options.

dgmulholland, 2025-08-13:
Good grief! Looks good though. I've ordered 2 new Pi 5s, heatsinks, and HATs to get my cluster started but I still need to get modules printed so I'm a bit behind you

closed 2025-09-13 by dgmulholland