JaredC01/LabStack #62: GL.iNet Comet

customade, 2025-08-04:
Hi, since we can't really get JetKVM in the US right now, I went down the [GL.iNet Comet](https://www.gl-inet.com/products/gl-rm1/) route. Could you add an adapter to work in one of the slots? I'll also see what I can do to try to modify one to work. It doesn't have screw holes for mounting so it will need to be braced a little better. Dimensions are 80mm long x 60mm wide x 18mm tall.

customade, 2025-08-13:
I took a crack at it in Fusion since I have one on hand. It fits really well. Super snug and doesn't move. I'll share the file. Though I probably need to make a left facing and a right facing port option. Also...this isn't as professional as yours so feel free to church it up.

JaredC01, 2025-08-13:
Looking forward to taking a poke at it!